PelionIoT/sd-driver #3: Cannnot change  the value of SDBlockDevice class members from outside

TomoYamanaka, 2017-03-31:
https://github.com/ARMmbed/sd-driver/blob/master/features/filesystem/sd/SDBlockDevice.h (line 130)
The access restriction of this class is **private:** .
For that reason, I can not change the value of this class members **from outside**.
<br>
For example, I would like to change `_transfer_sck` from outside by inheriting class. But I cann not change it. I would like to change the clock to about 20 MHz and access it fast.
Also, I would like to change other members as well.

In SDFilesystem of mbed classic, the access restriction of this class was **protected:** .
Therefore, I expect that the access restriction is chaged from **private:** to **protected:**.

Could you give us your opinion about the above ?
<br>

TomoYamanaka, 2017-04-17:
How is this going ?

cc @simonqhughes

geky, 2017-04-18:
A pr would be easier to move forward on

That being said, I'm not sure moving the internal variables of the class to protected is the best idea. This limits our ability to maintain the code, since we could no longer change those variables without breaking backwards compatibility.

Would changing the constructor to take in a `transfer_sck` be a reasonable solution?

What other members are you interested in changing? They may be better formed as getters/setters with appropriate documentation.

TomoYamanaka, 2017-04-21:
@geky 

Thank you for your reply.

> Would changing the constructor to take in a  transfer_sck  be a reasonable solution?

I guess that it is a reasonable solution. Even `SDFilesystem` that is the base of this library, the access restriction of this class was `protected:` .

> What other members are you interested in changing?

At least , [iSDIO](https://developer.mbed.org/users/ban4jp/code/iSDIO/) library inherits `SDFileSystem` and uses `_cmd` () `_cs` etc. 
Therefore , If `SDBlockDevice`  instead of `SDFileSystem` is used, `iSDIO` library will not work correctly.

deepikabhavnani, 2017-07-06:
@TomoYamanaka : Sorry for the delay. Frequency API is available in new SD driver. Let us know if that suffices your need.

TomoYamanaka, 2017-07-07:
@deepikabhavnani

Thank you for your comments.
I agree ragarding add the Frequency API. I will contact you if there is anything.

I close this issue.

closed 2017-07-07 by TomoYamanaka