PelionIoT/sd-driver #4: Sd-card initialization issue

yogpan01, 2017-04-03:
Init for the 1st time, initializing the modules
eSFS - test start
Unity test run 1 of 1
No disk, or could not put SD card in to SPI idle state
Fail to initialize card
<**UnityTest*>TEST(esfs, filename)</*UnityTest**>/home/danny/work/git/ex_mbed/mbed-client-esfs-example/esfs/Test/Unitest/esfs_test.c:106:TEST(esfs, filename):
===!!Unable to render embedded object: File (===> <***UnityResult***>FAIL</***UnityResult***> <===) not found.!!===: Expected 0 Was 4
Init: SDCARD_V2
init card = 1
SDHC Card
hc_c_size: 15159
capacity: 4310491136
sectors: 2305997493835366400
Additional file system operation may work

simonqhughes, 2017-04-03:
Thanks for logging this issue here. I have detailed information about this issue, which I will add here shortly.

simonqhughes, 2017-04-05:
I have a fix for this issue and will be submitted for review today.

JanneKiiskila, 2017-04-05:
@simonqhughes  - excellent, that's good news. @marcuschangarm seemed to have a card what was acting  worse than others, I think.  Would be interesting to get some card information that which cards are better or worse and why.

JanneKiiskila, 2017-04-06:
Simon's pull request is now available; https://github.com/ARMmbed/sd-driver/pull/5

simonqhughes, 2017-04-06:
The fix for this issue has now been merged and so the issue can be closed.

closed 2017-04-06 by simonqhughes