PelionIoT/sd-driver #14: UBLOX_ODIN_EVK_W2 - internal SD-card slot not working

JanneKiiskila, 2017-04-19:
With default mbed_lib.json the internal SD-card slot fails.

```
No disk, or could not put SD card in to SPI idle state
Fail to initialize card
```

Works with CI-shield, but why? Surely the internal SD-card slot should work as well.
There is no target override for UBLOX, tried adding one as well - but it still fails.

```
        "UBLOX_EVK_ODIN_W2": {
             "SPI_MOSI": "PE_14",
             "SPI_MISO": "PE_13",
             "SPI_CLK":  "PE_12",
             "SPI_CS":   "PE_11"
        },
```

Anyways, if look at the [pinout](https://developer.mbed.org/platforms/ublox-EVK-ODIN-W2/#pin-layout) for UBLOX, the pins are mapping to the CI-shield defaults - so it should work out of the box.

JanneKiiskila, 2017-04-19:
@bulislaw @simonqhughes  - any advice?

andreaslarssonublox, 2017-04-20:
When looking at the schematics for the ODIN EVK it looks like the CD/DAT3/CS pin is connected to D9. The CI-shield is using D10 for CS(according to arduino standard I think). We tested it during the workshop and it should work to just change the pin.
Maybe we should add a note to the ODIN EVK page @andreaspeterssonublox?

ODIN EVK schematics: https://developer.mbed.org/media/uploads/ECarlberg/evb-odin-w2_schematic.pdf
CI-shield schematics: https://github.com/ARMmbed/mbed-HDK/blob/master/Production%20Design%20Projects/CITestShield/v2.0.0/CITestShield%20V_2_0_0%20SCH.pdf

JanneKiiskila, 2017-04-20:
Now that you point it out, yes - it obvious (and I did take look at that spec as well, but still got it wrong). But, changing that one PIN makes all the difference now it works!  Thank you!

```
Starting example client

Init: SDCARD_V2
init card = 1

SDHC Card
hc_c_size: 7591
capacity: 7774208
sectors: 98248215760448
IPv4 mode
Start simple mbed Cloud Client
Resets storage to an empty state
```
I'll do a PR against this repo for this.

JanneKiiskila, 2017-04-20:
But, turns out we have another issue then. The Cloud Client will then later on fail, the SD-card file accesses start failing - but this issue is ONLY happening with the internal SD-card reader. Why? 

```
[DBG ][stor]: get_config_parameter
[INFO][mfcc]:  [ffffffff]key_config_manager.c:208:kcm_item_get_data: ===> kcm_item_name = mbed.UseBootstrap kcm_item_name_len = 17, kcm_data_der_max_size = 32

[DBG ][mfcc]:  [ffffffff]key_config_manager.c:36:kcm_add_prefix_to_name: ===> kcm_name_len=17

[DBG ][mfcc]:  [ffffffff]key_config_manager.c:52:kcm_add_prefix_to_name: <=== kcm_buffer_size_allocated_out=  26

[DBG ][mfcc]:  [ffffffff]storage.c:172:storage_file_read: ===> file_name_length=26, buffer_size=32

[DBG ][mfcc]:  [ffffffff]storage.c:133:storage_file_size_get: ===> file_name_length=26

[ERR ][esfs]: esfs_open errorExit res=0xfc000008

[ERR ][mfcc]:  [ffffffff]storage.c:142:storage_file_size_get: <=== Failed opening file (esfs_status 6)

[ERR ][mfcc]:  [ffffffff]storage.c:184:storage_file_read: <=== Failed getting file data size (kcm_status 10)

[ERR ][mfcc]:  [ffffffff]key_config_manager.c:223:kcm_item_get_data: Failed reading file from storage (10)

[ERR ][stor]: get_config_parameter kcm get error 10
```

JanneKiiskila, 2017-04-20:
PR created;  https://github.com/ARMmbed/sd-driver/pull/15

andreaspeterssonublox, 2017-04-20:
In the Pin Layout section at the ODIN-W2 EVK page it is already stated that the SD memory card connector is connected to D9, D11, D12 and D13. However I think it would be useful to add which SD memory card pin that maps to D9, D11, D12 and D13 respectively.

JanneKiiskila, 2017-04-21:
Is there any conflict with memory card and ethernet pins?

andreaspeterssonublox, 2017-04-24:
According to the ODIN-W2-EVK schematic there is no hw conflict with memory card and ethernet pins.

JanneKiiskila, 2017-05-04:
I might have been using the WiFi in that trial as well, so that could explain it. mbed-os issue - https://github.com/ARMmbed/mbed-os/issues/4257 - was raised about that and you already diagnosed that with good speed, thanks for that.

closed 2017-05-04 by JanneKiiskila