PelionIoT/sd-driver #19: Comilation failure

akrnvm, 2017-05-11:
i followed the above link, still i got slimier issue,

[Warning] SDBlockDevice.h@122,0: #997-D: function "BlockDevice::size()" is hidden by "SDBlockDevice::size" - virtual function override intended? [Error] example1.cpp@11,0: #322: object of abstract class type "SDBlockDevice" is not allowed: [ERROR] "./features/filesystem/sd/SDBlockDevice.h", line 122: Warning: #997-D: function "BlockDevice::size()" is hidden by "SDBlockDevice::size" - virtual function override intended? ".\example1.cpp", line 11: Error: #322: object of abstract class type "SDBlockDevice" is not allowed: pure virtual function "BlockDevice::size" has no overrider .\example1.cpp: 1 warning, 1 error

steps: 1. mbed OS v5.4 2. SD driver from https://github.com/ARMmbed/sd-driver/ 3. placed the driver in filesystem 4. Compiled expample1.cpp (is provied)mbed compile -t ARM -m XXXX

instead of SD Block system, if i used SD Filesystem Compilation is ok, but card detection is Fail.

0xc0170, 2017-05-11:
@akrnvm reproduced on more targets as XXX is referenced above via -m ?

JanneKiiskila, 2017-05-15:
@akrnvm -  yep, exact board details & way how you connected the SD-card is vital information.

akrnvm, 2017-05-16:
Hi, 
using Filesystem object i'm able to communicate with SD card.

fopen () and fread() commands are not working, f_open,f_write are working.

Not Working Code:
Ref: https://developer.mbed.org/cookbook/SD-Card-File-System

#include "mbed.h"
#include "SDFileSystem.h"
 
SDBlockDevice sd(MOSI, MISO, SCLK, PJ1);
FATFileSystem fs("sd", &sd);

int main() {
    mkdir("/sd/mydir", 0777);
    
    FILE *fp = fopen("/sd/mydir/sdtest.txt", "w");
    if(fp == NULL) {
        error("Could not open file for write\n");
    }
    fprintf(fp, "Hello fun SD Card World!");
    fclose(fp); 
}

Working Code:

SDBlockDevice sd(MOSI, MISO, SCLK, PJ1);
FATFileSystem fs("sd", &sd);
int main() {
	FATFS FatFs;
	FIL pFilePtr;
        f_mount(&FatFs, " ",1);
	f_open(&pFilePtr, "0:/URC/CONFIG/WebResponse1.txt", FA_WRITE  FA_OPEN_ALWAYS);
        f_write(&pFilePtr, "test2", 5, &uiValue);
	f_close(&pFilePtr);
}

geky, 2017-05-16:
Hi @akrnvm, you are not going crazy, that code should work. I'll see if I can reproduce this.

Is this on the [LPC1768](https://developer.mbed.org/platforms/mbed-LPC1768/)?

akrnvm, 2017-05-18:
>Is this on the LPC1768?
No,its our own board with 16KB RAM.
could you please explain the logical drive "sd" and "0:" difference ?, why can't its open with "sd/URC/CONFIG/WebResponse1.txt" .

akrnvm, 2017-05-19:
hi,
could you please tell, is it possible to execute SD block device application on 16KB RAM device?
my code is working with Keil but its Fail in IAR.
is IAR required more RAM ?

0xc0170, 2017-05-19:
> my code is working with Keil but its Fail in IAR.

What's the failure? Does not fit in the RAM region? using debug and even release profile?

deepikabhavnani, 2017-11-20:
@akrnvm - Not sure if this issue still exists, all test in SD repo are successful with IAR 7.8 and higher versions and latest mbed-os version. Please re-open if required.

closed 2017-11-20 by deepikabhavnani