PelionIoT/sd-driver #23: heap allocation error after using sd-driver

masoudr, 2017-06-06:
Hi,
I'm using this library with mbed-connector to read and write sd files with the cloud and my platform is K64F.
I'm not sure if my problem is related to this library but I tried to remove the sd read/write and everything worked fine. Here is my problem:
After compiling and running the code it works fine but after a few minute, I got the error "Operator new out of memory" in the terminal. I'm using these two functions to read and write to sd. (I removed the other part of code for)

```
#include "mbed.h"
#include "security.h"
#include "easy-connect.h"
#include "simple-mbed-client.h"
#include <stdio.h>
#include <cstdlib>
#include "C12832.h"
#include "LM75B.h"
#include "MMA7660.h"
#include <cstring>
#include <string>
#include <sstream>  //for conversion of data types to string
#include <iomanip> //for setting precision of float
#include "FATFileSystem.h"
#include "SDBlockDevice.h"
#include <errno.h>
/* mbed_retarget.h is included after errno.h so symbols are mapped to
* consistent values for all toolchains */
#include "platform/mbed_retarget.h"

SimpleResourceString SDcard_write = client.define_resource("device/0/SD_W", "RED", M2MBase::GET_PUT_ALLOWED, true, &SD_write_updated);
SimpleResourceString SDcard_read = client.define_resource("device/0/SD_R", "RED", M2MBase::GET_ALLOWED, true);

void sd_write_open()
{
  string str2;
  //  char *y;
  //  int size=128;
  //  char *y =(char *)malloc(size);
  string sub="\\n"; // str2 is string to search, sub is the substring to search for
  vector<size_t> positions; // holds all the positions that sub occurs within str
  size_t pos = sd_write_value.find(sub, 0);
  while(pos != string::npos) {
    positions.push_back(pos);
    pos = sd_write_value.find(sub,pos+1);
  }
  for(int i=0; i<positions.size(); ++i) {
    printf("list the posistions %d\n",positions[i]);
  }
  //    printf("sd_write_value size is %d\n",sd_write_value.size());
  printf("Writing to Sd card...\n");
  FILE* fd = fopen("/sd/mydir/out.txt", "w+");
  errno_error(fd);
  for(int i=0; i<=positions.size(); i++) {
    if(i==0) {
      str2= sd_write_value.substr(0,positions[i]);
    } else if (i==positions.size()) {
      str2= sd_write_value.substr(positions[i-1]+2,sd_write_value.size());
    } else {
      str2= sd_write_value.substr(positions[i-1]+2,positions[i]-(positions[i-1]+2));
    }
    //  y=str2.c_str();
    //    char *y = &str2[0u];
    printf("here is the %d line: ",i);
    printf("%s\n",str2);
    fprintf(fd, "%s\n", str2);
  }
  fclose(fd);
  printf("Done");
}
void sd_read_open()
{
  string sum;
  printf("Reading file at /sd/mydir/in.txt\n");
  FILE *fp = fopen("/sd/mydir/in.txt", "r");
  char buffer [128];
  size_t i=0;
  while (!feof(fp)){
    bool ret =  (fgets(buffer, 64, fp)) ;
    if(ret)
    {
      sum.append(buffer);
      printf("Here is the %d line:",i);
      printf ("%s\n", buffer);
    }
    i++;
  }
  SDcard_read=sum;
  //  printf("Here is the sum: \n");
  //  printf("%s\n",sum);
  fclose(fp);
}
```
Thanks.

deepikabhavnani, 2017-08-04:
@masoudr - Sorry for delayed response. SD driver was updated recently. May I know which version of driver was having this issue. Also, if possible can you verify the behavior with updated SD driver (sd-driver-0.1.1-mbed-os-5.5.4) and mbed-os (mbed-os-5.5.4) and share your findings?

masoudr, 2017-08-04:
Hi, sorry I forgot to respond. No, the problem solved because actually it was not related to `sd-driver`. I tested the above program with mbed-os 5.5.4 (I can't remember the version of sd-driver) and it was ok for reading and writing to text files. Currently, I'cant reach my device but I will test it and report if it runs into problem.

deepikabhavnani, 2017-08-04:
Thanks for responding. I will close the issue for now. Feel free to open it, if required.

closed 2017-08-04 by deepikabhavnani