PelionIoT/sd-driver #39: All PAL filesystem tests fail

avolinski, 2017-07-09:
Hi,

PAL File System tests fail on all mbedOS running devices after changes introduced several days ago into sd-driver repository.
https://jenkins-internal.mbed.com/view/PAL/job/PAL-Tests-Pipeline-Nightly/306/testReport/

was there some basic change in the init behavior?

deepikabhavnani, 2017-07-10:
Yes.. init behavior was changed and instead polling, timeouts for command was implemented.
@avolinski - May i know which SD card is used for PAL ?

If possible can you send me SD Command trace for any failing test by enabling
'SD_DBG'  and 'SD_CMD_TRACE' flag in SDBlockDevice.cpp.

netanelgonen, 2017-07-11:
[sd-logError.txt](https://github.com/ARMmbed/sd-driver/files/1138546/sd-logError.txt)
@deepikabhavnani please see attched file
im using transcend 8 gb micro sd hc class 10

netanelgonen, 2017-07-11:
@galc @JanneKiiskila

JanneKiiskila, 2017-07-11:
Are you calling the `sd.init()` in the pal tests?

netanelgonen, 2017-07-11:
@JanneKiiskila actually no...
following this https://github.com/ARMmbed/sd-driver/blob/master/TESTS/filesystem/basic/basic.cpp example i have not seen any use of it...

i have added the function but the errors persist, send a new log?(the errors are the same...)

JanneKiiskila, 2017-07-11:
Well, if you did do it - and caught an error there, you could immediately stop all file tests.

netanelgonen, 2017-07-11:
sd.init returns 0

netanelgonen, 2017-07-12:
@deepikabhavnani is there any update with this? 

this actually blocks all our file system tests

JanneKiiskila, 2017-07-12:
@netanelgonen  - you could try the PR; https://github.com/ARMmbed/sd-driver/pull/41

deepikabhavnani, 2017-07-12:
@netanelgonen - Did you get the chance to test PAL test with #41 fix? 

After above fix, I have seen SPI block read failures with few SD cards, and fixing that might take sometime in mbed-os. https://github.com/ARMmbed/mbed-os/pull/4744

Quick temporary solution to that would be to use SPI byte read.

netanelgonen, 2017-07-13:
@deepikabhavnani #41 did not solve the problem

deepikabhavnani, 2017-07-13:
Does it still fail in initialisation?

JanneKiiskila, 2017-07-13:
Is this the root cause for the issue now? https://github.com/ARMmbed/mbed-os/issues/4743 
Or is it something else?

JanneKiiskila, 2017-07-17:
Can we get some updates on this? @netanelgonen  - can you post latest traces?

netanelgonen, 2017-07-17:
@JanneKiiskila unfortunately the PR is not solving the issue and the traces are them same..
if there is a need for the new trace i will upload them later today.

deepikabhavnani, 2017-07-17:
Will it be possible to share SPI bus trace?

netanelgonen, 2017-07-18:
please view the attched file
[sdlog.txt](https://github.com/ARMmbed/sd-driver/files/1155083/sdlog.txt)

deepikabhavnani, 2017-07-18:
Initialisation is successful from log, do you get the return (0 -success) from init? Not sure why re-init is done?

Card CSD in this log was success and got 8GB card size

netanelgonen, 2017-07-18:
the init is successful but we cannot read/write from the sd card...

netanelgonen, 2017-07-18:
we encountering inconsistent behavior with the new sd-driver,
sometime all test fails, sometimes all pass, and sometimes half pass and half fail... 
it may be timing issues... 
this is a link to our jenkins job: 
https://jenkins-internal.mbed.com/view/PAL/job/PAL-Tests-Pipeline-Nightly/317/testReport/

the file system tests start failing after your update.
in meanwhile do you have a tag before the update that we can use?

deepikabhavnani, 2017-07-18:
PAL failures are with K64F devices, which is related to https://github.com/ARMmbed/mbed-os/issues/4743, SPI block reads fail or K64F. 

For older versions and tag please refer to readme, or use  'git tag -n' command in driver folder

avolinski, 2017-08-02:
one small note, originally pal failures were not only with k64f devices, but with odin boards too.
so failures are not bound to one platform
https://jenkins-internal.mbed.com/view/PAL/job/PAL-Tests-Pipeline-Nightly/306/testReport/

deepikabhavnani, 2017-08-04:
@avolinski @netanelgonen @JanneKiiskila - I am closing this issue. If PAL failures related to SD device are found, please open new issue.

closed 2017-08-04 by deepikabhavnani