PelionIoT/sd-driver #46: Getting error with STM32L432KC

mcprakash, 2017-08-09:
Hi
 I am trying to use the mbed-os-example-sd-driver from here -> https://github.com/ARMmbed/mbed-os-example-sd-driver/blob/master/main.cpp with my STM32L432KC. I can compile fine and am able to run it. PFA my output.

[putty.txt](https://github.com/ARMmbed/sd-driver/files/1213093/putty.txt)

Also, I slightly changed the code to match the pinout on the SD card device I am using. I am using a CATALEX sd card module with a Sandisk 16 GB sd card. A schematic is shown below 
[image]

Can you help me sort out the problem ? Thanks

Best
Chandra

deepikabhavnani, 2017-08-10:
File open for the card is failing. Looks like SD card is not initialised properly. Can you share version of sd driver and mbed-os.

mcprakash, 2017-08-10:
I tried to use the standard steps described at *mbed-os-example-sd-driver*

mbed-os -> commit f4864dc6429e1ff5474111d4e0f6bee36a759b1c
sd-driver -> commit 96142e4303df6540b4b231453dd23701111dc1bb

deepikabhavnani, 2017-08-10:
Sorry for the inconvenience, but example driver is not updated with latest fixes for sd-driver and mbed-os.
Please use the latest release tag sd-driver-0.1.1-mbed-os-5.5.4 and mbed-os-5.5.4

mcprakash, 2017-08-10:
Hi @deepikabhavnani 
 Apologies for troubling again. Could you let me know the latest branch or commit (specifically for mbed-os) which I can pull. I think _**sd-driver-master**_ is already upto-date. 

 I tried with the latest commit in branch **_mbed-os-5.5_** (from repository **_mbed-os_**) and got the following result. This time it is **error 5**. 

 Attached debug output
[putty.txt](https://github.com/ARMmbed/sd-driver/files/1215974/putty.txt)

 Thanks 
 Chandra

deepikabhavnani, 2017-08-10:
Yes, you will need [mbed-os-5.5.4](https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.5.4)

mcprakash, 2017-08-10:
I tried with mbed-os-5.5.4 branch. PFA snapshots of the git revisions
[image]
[image]

I am still getting the same **_error failure 5_**. Let me know if you would like me to format the SD card and try again. It is in FAT32 format. 

Thanks
Chandra

deepikabhavnani, 2017-08-10:
Source and connection looks fine, just verify correct selection is done for PB_4 pin (shared between I2C/SPI).
Also please enable below flags in SDBlockDevice.cpp and share the serial log.

```
#define SD_DBG                                   0      /*!< 1 - Enable debugging */
#define SD_CMD_TRACE                             0      /*!< 1 - Enable SD command tracing */
```

mcprakash, 2017-08-10:
PFA the output with the DEBUG option enabled

[putty.txt](https://github.com/ARMmbed/sd-driver/files/1216521/putty.txt)

Thanks
Chandra

deepikabhavnani, 2017-08-11:
As per log, there is no response from the card. Also, SD cards operate either in 3.3V-2.7V range or low power 1.8V. Are you sure of 5V?
Please verify the voltage, as well pin mux for PA11 / PB_3 / PB_4.

mcprakash, 2017-08-11:
Hi @deepikabhavnani 
 I have some updates from my end. The mbed-2 library FATFileSystem works with the same configuration, pinout and setup. PFA some of my results. 
 
 
[image]
[image]
[image]

So, I know for sure that I have done everything correctly. Now, moving on to SDBlockDevice.h, for this I was using the offline mbed compile -t NUCLEO_L432KC -m GCC_ARM with the MINGW compiler for windows. Never had any issue with it, but would like to confirm if it is supported. If not, could you guide me into importing the sample example to mbed online compiler and trying it out. 

Thanks 
Chandra

deepikabhavnani, 2017-08-11:
As per log from example driver we didn't see any response from device. Will it be possible to connect logic analyser on spi bus and capture trace?

deepikabhavnani, 2017-08-12:
I found issue in SD-driver, specific to Sandisk card (tried 64GB). Added fix in PR #48. Please try and let me know if that works for you or not.

mcprakash, 2017-08-12:
HI @deepikabhavnani 
 Apologies for the slightly delayed reply. I just tested it out. I am still seeing some issues, particularly that this library is pretty slow as compared to **_SDFileSystem_**. It may also be because of all the error codes and wait times. Maybe adding a slight delay to these particular commands might help. 

PFA the debug output
[putty.txt](https://github.com/ARMmbed/sd-driver/files/1219825/putty.txt)

Thanks again for helping me out. 

Best
Chandra

deepikabhavnani, 2017-08-12:
I can have init frequency as 400khz. You can increase transfer frequency and verify for CMD17.
Also make sure to disable debugs :-)

deepikabhavnani, 2017-08-12:
Also, SD cards should always be flashed with SD format utility and not with windows/Linux to have good performance. Can you give that a try. If still it takes 4/5 sec to initialise than i will have to debug.

mcprakash, 2017-08-12:
Hi @deepikabhavnani 
 Thank you for figuring it out. Your fix does work but it looks really slow especially at CMD:17 at the start of the program (typically 1 min). 

 PFA the debug output from one of the runs 
[putty.txt](https://github.com/ARMmbed/sd-driver/files/1219861/putty.txt)

 Also, I think the de-initialization has some errors because if I press the RESET switch, I am getting errors (a.k.a. it works the first time, but on reboot somehow it does not work well

Thanks
Chandra

mcprakash, 2017-08-12:
Sure, I will try and let you know by Monday. Thanks

mcprakash, 2017-08-15:
Hi @deepikabhavnani 
 Thank you for helping me with the SD card. I can confirm that it is very fast and is taking less than a second. 
 
Best
Chandra

closed 2017-08-28 by deepikabhavnani