PelionIoT/sd-driver #51: Timing issues

LemonBoy, 2017-08-14:
Long story short, this microsd card is pretty slow (and tiny) and am having a hard time getting it to play along with the sd-driver (with #48, #49, #50).
For some reason I keep hitting some weird IO error that I haven't been able to pinpoint yet as every time I add a printf to log some values the bug disappears and the program is able to reach the end thanks to the delay it introduces.
Here's a short log of the commands, I haven't been able to track down what command causes this so I hope you may have any hint.

```
CMD:24 	 arg:0x141a00 	 Response:0x0 
CMD:13 	 arg:0x0 	 Response:0x0 
R2: 0x0 
CMD:23 	 arg:0x2 	 Response:0x0 
CMD:25 	 arg:0x141c00 	 Response:0x0 
CMD:17 	 arg:0x100800 	 Response:0x0 
_wait_token: timeout
Read timeout
CMD:17 	 arg:0x100800 	 Response:0x0 
CMD:23 	 arg:0x4 	 Response:0x0 
CMD:25 	 arg:0x144000 	 Response:0x0 
CMD:24 	 arg:0x144800 	 Response:0x0 
CMD:13 	 arg:0x0 	 Response:0x0 
R2: 0x0 
CMD:24 	 arg:0x100800 	 Response:0x0 
No response CMD:13 
No response CMD:13 
No response CMD:13 
No response CMD:13 

No response CMD:24 
No response CMD:24 
No response CMD:24 
No response CMD:24 
```

LemonBoy, 2017-08-15:
After some more digging I've noticed that the problem is connected to the use of `CMD25`, I've been able to get the whole program to work reliably by replacing its use with multiple `CMD24` at the expense of the write troughtput. I think some more syncing is needed somewhere.

```
// trying to write a single 2048 byte block
CMD:23 	 arg:0x4 	 Response:0x0 
CMD:25 	 arg:0x143000 	 Response:0x0 
resp 5 4 2048
resp 5 3 2048
resp 5 2 2048
resp 5 1 2048
// the card is dead from now on, _every_ command is illegal
Illegal command CMD:23
```

deepikabhavnani, 2017-08-16:
@LemonBoy - If the card is slow, I would recommend using CMD24 instead CMD25, as that is for perfomance.
Also, for timing critical issue it will be helpful if you can share logic analyzer trace with CMD25 vs CMD24. Debug prints won;t help so can have them disabled.

LemonBoy, 2017-08-17:
> If the card is slow, I would recommend using CMD24 instead CMD25, as that is for perfomance.

Sure, but I think the intermittent `CMD25` failures should be treated as a bug.

> Also, for timing critical issue it will be helpful if you can share logic analyzer trace with CMD25 vs CMD24. 

I don't have access to a logic analyzer at the moment, that's why I tried to debug this problem with the tools at hand but ran out of time.

> Also please share the card details, I will see if i can collect one similar one.

It is a no-name 128Mb card that was probably sourced from some chinese website, I think you can easily find some equally shitty ones :)

deepikabhavnani, 2017-08-30:
Please check if PR #59 resolves CMD25 timing issue for you. sorry I still don't have 128Mb card so cant test or debug.

deepikabhavnani, 2017-09-11:
@LemonBoy - Please, let us know if timing issue still exists with master branch.

LemonBoy, 2017-09-28:
Sorry for the delay, I've just tested the master branch and it works just fine.
Thanks for looking into this!

closed 2017-09-28 by LemonBoy