PelionIoT/sd-driver #62: Can't compile against mbed-OS master

JanneKiiskila, 2017-09-18:
One can't compile against the mbed OS 5.6.0 RC1.

```
09:47:56 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] Compile [ 97.6%]: SDBlockDevice.cpp
09:47:56 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] [Error] SDBlockDevice.cpp@152,3: [Pe018]: expected a ")"
09:47:56 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] [Fatal Error] SDBlockDevice.cpp@153,0: [Pe035]: #error directive: "Incompatible mbed-os version detected! Required 5.5.4 and above"
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] [ERROR] 
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR]   #if (MBED_VERSION < MBED_ENCODE_VERSION(5,5,4))
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR]        ^
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] "C:\builds\ws\app_PR-545-PPVBV4HCVIU2SIBUWGKPS\mbed-client-testapp\sd-driver\SDBlockDevice.cpp",152  Error[Pe018]: expected a ")"
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] 
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR]   #error "Incompatible mbed-os version detected! Required 5.5.4 and above"
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR]    ^
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] "C:\builds\ws\app_PR-545-PPVBV4HCVIU2SIBUWGKPS\mbed-client-testapp\sd-driver\SDBlockDevice.cpp",153  Fatal error[Pe035]: #error directive: "Incompatible mbed-os version detected! Required 5.5.4 and above"
09:47:57 [UBLOX_EVK_ODIN_W2_LWIP4 IAR] Fatal error detected, aborting.
```

This needs to be fixed ASAP.

JanneKiiskila, 2017-09-18:
Passes with GCC. Seems this is sort of an IAR issue. But, do we REALLY need this check?

JanneKiiskila, 2017-09-18:
Seems to be an issue with `mbed.h` itself, line 35 is wrong.

```
#define MBED_VERSION MBED_ENCODE_VERSION(MBED_MAJOR_VERSION, MBED_MINOR_VERSION, MBED_PATCH_VERSION)        #define MBED_H
```
This extra `#define MBED_H` should not really be there! @adbridge @0xc0170

JanneKiiskila, 2017-09-18:
https://github.com/ARMmbed/mbed-os/pull/5132 created

deepikabhavnani, 2017-09-18:
> Passes with GCC. Seems this is sort of an IAR issue. But, do we REALLY need this check?

Check was added so that compatible version of mbed-os is picked with sd-driver. Previously any combination was picked up and we use to get issues because of incompatibility. (example Block read/write new API's were added which won;t be available with older mbed-os version).

JanneKiiskila, 2017-09-18:
I think this issue can be closed, it's really an mbed OS issue rather than sd-driver issue.

closed 2017-09-18 by JanneKiiskila