PelionIoT/sd-driver #79: Intermittent Data taking too long

mcprakash, 2018-01-10:
Hi @deepikabhavnani 
 I am trying to write sensor data to SD card (approximately 150 bytes every 10 ms). The typical write time is less than 1 ms, but it is not uniform. With both DBG and CMD_Trace on, as you can see after about 5/6 cycles, I am getting a delay of approx 20ms (with debug print on) and about 5 ms without debug print. 
 
 Please do help me out in resolving this issue. I am using mbed-os-5.5.4 and sd-driver-0.1.1 as recommended. 
[log_sd.txt](https://github.com/ARMmbed/sd-driver/files/1617450/log_sd.txt)

deepikabhavnani, 2018-01-10:
Thanks for the log file. Erase command for single block is failing, as per the log. In case of SD cards erase prior to every block write is not required, hence erase was moved to trim. Please use the latest version ‘sd-driver-0.1.2-mbed-os-5.6.1’ and let me know if issue still exists.

mcprakash, 2018-01-10:
Hi @deepikabhavnani 
 I tried with sd-driver-0.1.2 and mbed-os-5.6.1. Now, it is slightly better(the delay is shorter 1 ms w/o debug print and 10 ms with debug print), but the problem is still unresolved.

 PFA the log with debug print 
[putty.log](https://github.com/ARMmbed/sd-driver/files/1617557/putty.log)
  and the one without debug print 
 
[log_sd.txt](https://github.com/ARMmbed/sd-driver/files/1617558/log_sd.txt)

 Please do let me know if commenting out the trim would help. 

Best
Chandra

deepikabhavnani, 2018-01-10:
No need to disable trim, as it is performed only during format. I see sequence of CM23 / CMD25 / CMD24. Which is for multiple block writes. It seems application is trying to write 1K of data (2 blocks) and not 150 bytes. Can you cross check your application, or please share the code snippet to reproduce this.

mcprakash, 2018-01-10:
I am writing 2 data structs every 10ms. 

```
typedef struct
{
        char ch_enc1[4];
        int t;
        float q[4];
        float flat;
        float flon;
        float falt;
        float pos[3];
        float vel[3];
        int radar_dist;
} nav_struct;

typedef struct
{
        float acc[3];
        float gyro[3];
        float mag[3];
        float acc_filt[3];
        float gyro_filt[3];
        float mag_filt[3];
}imu_struct;

```

The main function looks like this 

```
while(1)
	{
		nav_data.val.t = t.read_ms();

		// Update sensors
		update_imu();
		update_radar();

		// Run AHRS 
		run_ekf();
		
		// Store data
	#if DATA_STORE_SD
		sd_write_nav_data();
		sd_write_imu_data();
	#endif

		// Stop FCS if user pressed stop
		check_stop_condition();

		// Wait till DT time elapses
		t_delay = DT - (t.read_ms() - nav_data.val.t);
		wait_ms(t_delay>0?t_delay:0);
		radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
	}

```

deepikabhavnani, 2018-01-10:
sd_write_nav_data();
		sd_write_imu_data();
More on these functions please. How is SD driver used to perform write? How big is the buffer?

mcprakash, 2018-01-10:
I have edited the above code to give a much better idea

The functions are listed below:

```
void sd_write_imu_data()
{
      #if SD_BINARY
            fwrite(imu_data.ch, sizeof(imu_data.ch), 1, fp);
      #else
            sprintf(ch, "{\"t\": %d, \"acc\": [%f, %f, %f], \"gyro\": [%f, %f, %f], \"mag\": [%f, %f, %f], \"acc_filt\": [%f, %f, %f], \"gyro_filt\": [%f, %f, %f], \"mag_filt\": [%f, %f, %f]} \r\n", nav_data.val.t, imu_data.val.acc[0], imu_data.val.acc[1],
                              imu_data.val.acc[2], imu_data.val.gyro[0], imu_data.val.gyro[1], imu_data.val.gyro[2], imu_data.val.mag[0], imu_data.val.mag[1], imu_data.val.mag[2],  imu_data.val.acc_filt[0], imu_data.val.acc_filt[1], imu_data.val.acc_filt[2],
                              imu_data.val.gyro_filt[0], imu_data.val.gyro_filt[1], imu_data.val.gyro_filt[2], imu_data.val.mag_filt[0], imu_data.val.mag_filt[1], imu_data.val.mag_filt[2]);
            fputs(ch, fp);
      #endif
}

void sd_write_nav_data()
{
      #if SD_BINARY
            fwrite(nav_data.ch, sizeof(nav_data.ch), 1, fp);
      #else
            sprintf(ch, "{\"t\": %d, \"q\": [%f, %f, %f, %f], \"flat\": %f, \"flon\": %f, \"falt\": %f, \"pos\": [%f, %f, %f], \"vel\": [%f, %f, %f], \"radar_dist\":%d } \r\n", nav_data.val.t, nav_data.val.q[0], nav_data.val.q[1], nav_data.val.q[2], nav_data.val.q[3], nav_data.val.flat, nav_data.val.flon, nav_data.val.falt, nav_data.val.pos[0], nav_data.val.pos[1], nav_data.val.pos[2], nav_data.val.vel[0], nav_data.val.vel[1], nav_data.val.vel[1], nav_data.val.radar_dist);
            fputs(ch, fp);
      #endif
}
```

deepikabhavnani, 2018-01-10:
`char ch[sizeof(imu_struct)];`  - Can you print size of this array.
`sprintf(ch, "{\"t\": %d, \"acc\": [%f, %f, %f], \"gyro\` - Seems some global can you list the size of this?

Update the timing code
```
while(1)
	{
		--nav_data.val.t = t.read_ms();

		// Update sensors
		update_imu();
		update_radar();

		// Run AHRS 
		run_ekf();
++nav_data.val.t = t.read_ms();
		// Store data
	#if DATA_STORE_SD
		sd_write_nav_data();
		sd_write_imu_data();
	#endif
++t_delay = (t.read_ms() - nav_data.val.t);
		// Stop FCS if user pressed stop
		check_stop_condition();

		// Wait till DT time elapses
		--t_delay = DT - (t.read_ms() - nav_data.val.t);
		--wait_ms(t_delay>0?t_delay:0);
		radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
               ++t_delay = DT - (t.read_ms() - nav_data.val.t);
                ++wait_ms(t_delay>0?t_delay:0);
	}

```

deepikabhavnani, 2018-01-10:
Sorry for indentation

mcprakash, 2018-01-10:
Hi @deepikabhavnani 
 As seen on the ARM device using the following code 

```
radio.printf("imu_size = %d \r\n", sizeof(imu_data.ch));
radio.printf("nav_size = %d \r\n", sizeof(nav_data.ch));
```
```
imu_size = 72 
nav_size = 64 
```

The sprintf("...") is for json printing which is not being used. 

The total write size for each cycle is 136 bytes. 

Best
Chandra

deepikabhavnani, 2018-01-10:
Can you see the timing with the code changes as described in previous comment

mcprakash, 2018-01-10:
Hi @deepikabhavnani 
With the updated code

```
while(1)
	{
		nav_data.val.t = t.read_ms();

		// Update sensors
		//update_imu();
		//update_radar();

		// Run AHRS 
		//run_ekf();
		
		// Store data
	#if DATA_STORE_SD
		sd_write_nav_data();
		sd_write_imu_data();
	#endif

		// Stop FCS if user pressed stop
		check_stop_condition();

		// Wait till DT time elapses
		t_delay = DT - (t.read_ms() - nav_data.val.t);
		radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
		wait_ms(t_delay>0?t_delay:0);
	}
```

The log file is attached
[putty.log](https://github.com/ARMmbed/sd-driver/files/1620400/putty.log)

deepikabhavnani, 2018-01-10:
check_stop_condition(); - Can you comment that as well..  Also, if possible remove `wait_ms()`

deepikabhavnani, 2018-01-10:
What I see is not the SD command processing time taking time, but the wait.
```
		sd_write_nav_data();
		sd_write_imu_data();

**CMD:17 	 arg:0x10c00 	 Response:0x0 
CMD:23 	 arg:0x2 	 Response:0x0 
CMD:25 	 arg:0x90000 	 Response:0x0 
CMD:24 	 arg:0x10c00 	 Response:0x0 
CMD:24 	 arg:0x2e600 	 Response:0x0 
CMD:17 	 arg:0x50000 	 Response:0x0 
CMD:24 	 arg:0x50000 	 Response:0x0 **
```
radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
10 	 10 	 10

wait_ms(t_delay>0?t_delay:0); // Here you wait for 10 ms, before updating next data.

mcprakash, 2018-01-10:
The updated code is as follows 

```
while(1)
	{
		nav_data.val.t = t.read_ms();

		// Update sensors
		//update_imu();
		//update_radar();

		// Run AHRS 
		//run_ekf();
		
		// Store data
	#if DATA_STORE_SD
		sd_write_nav_data();
		sd_write_imu_data();
	#endif

		// Stop FCS if user pressed stop
		//check_stop_condition();

		// Wait till DT time elapses
		t_delay = DT - (t.read_ms() - nav_data.val.t);
		radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
		//wait_ms(t_delay>0?t_delay:0);
	}
```

The log file is attached 
[putty.log](https://github.com/ARMmbed/sd-driver/files/1620418/putty.log)

The sd card being used is 
https://drive.google.com/file/d/1wZT1Nq_CdqTahKMeVLWQWJilte-pHEjh/view?usp=sharing

mcprakash, 2018-01-10:
You might be mistaken, I wait to make sure that the cycle is at least 10ms and not 10ms for every cycle. \

Best
Chandra

deepikabhavnani, 2018-01-10:
```
while(1)
	{
		nav_data.val.t = t.read_ms();
	#if DATA_STORE_SD
		sd_write_nav_data();           - Does write always and no condition
		sd_write_imu_data();

CMD:23 	 arg:0x2 	 Response:0x0 
CMD:25 	 arg:0x91c00 	 Response:0x0 
CMD:24 	 arg:0x50000 	 Response:0x0 

	#endif
		t_delay = DT - (t.read_ms() - nav_data.val.t);
		radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
I dont see additional loop or anything over here.. How come you are getting multiple print statements?
201 	 184 	 -7
202 	 202 	 10
203 	 203 	 10
205 	 205 	 10
206 	 206 	 10
207 	 207 	 10
209 	 209 	 10
210 	 210 	 10
Is there some other thread? We should have just one print with varying data value
	}
```

mcprakash, 2018-01-10:
Since, the wait_ms() function is now commented. The code just tries to write as fast as possible. 

The t_delay is now only a dummy value. (Always 10) indicating the sd_write() takes less than 1 ms in ideal conditions. 

except for when the CMD:23/24/25 takes place. 

Hope it makes more sense

deepikabhavnani, 2018-01-10:
I understand the write is not taking time, what i don't understand multiple prints 
201 	 184 	 -7
202 	 202 	 10
203 	 203 	 10
205 	 205 	 10
206 	 206 	 10
207 	 207 	 10
209 	 209 	 10
210 	 210 	 10

mcprakash, 2018-01-10:
Every cycle you print the values for that particular iteration as it is a while(1) loop

mcprakash, 2018-01-10:
so only if wait_ms(t_delay) is present, will these values be differing by 10

201 184 -7   // say iteration 1
202 202 10  // say iteration 2
203 203 10  // ...
205 205 10
206 206 10
207 207 10
209 209 10
210 210 10 // say iteration 10

deepikabhavnani, 2018-01-10:
There is write in between that iteration, your code is  not matching the prints, do you have additional thread printing something?
Its should be 
SD write (CMD)
print timing
SD write (CMD)
print timing

mcprakash, 2018-01-10:
I did not understand.

Do you mean I should see something like this

```
CMD:23
xxx yyy zzz
CMD:24
xxx yyy zzz
CMD:25
xxx yyy zzz

```
Nope, it is a single thread application

deepikabhavnani, 2018-01-10:
```
nav_data.val.t = t.read_ms();
sd_write_nav_data();
sd_write_imu_data();
t_delay = DT - (t.read_ms() - nav_data.val.t);
radio.printf("%d \t %d \t %d\r\n", t.read_ms(), nav_data.val.t, t_delay);
```
This is single iteration. 

No CMD23/CM24/CMD25 will be in sequence for sd_write.
'''
CMD:23 	 arg:0x2 	 Response:0x0 
CMD:25 	 arg:0x90400 	 Response:0x0 
CMD:24 	 arg:0x50000 	 Response:0x0 
70 	 70 	 10
CMD:23 	 arg:0x2 	 Response:0x0 
CMD:25 	 arg:0x90400 	 Response:0x0 
CMD:24 	 arg:0x50000 	 Response:0x0 
96 	 96 	 10
''' 
Is what i meant..

deepikabhavnani, 2018-01-10:
Looking at log, for timing first print always has -ve delay
44 	 6 	 -28
69 	 53 	 -6
95 	 78 	 -7
147 	 131 	 -6

mcprakash, 2018-01-10:
Not sure why the first few statements always have -ve delay. 

But, I can say this 

CMD:23 arg:0x2 Response:0x0
CMD:25 arg:0x90400 Response:0x0
CMD:24 arg:0x50000 Response:0x0
70 70 -6 // This iteration had a total delay of 16 ms which is greater than ideal delay of 10ms
71 71 10 // This iteration worked fine
xxx yyy zzz // This iteration worked fine
CMD:23 arg:0x2 Response:0x0
CMD:25 arg:0x90400 Response:0x0
CMD:24 arg:0x50000 Response:0x0
96 96 10

deepikabhavnani, 2018-01-10:
Can you try this 
```
int final_time = 0;
while(1) {
nav_data.val.t = t.read_ms();
sd_write_nav_data();
sd_write_imu_data();
final_time = t.read_ms();
t_delay = DT - (final_time - nav_data.val.t);
radio.printf("%d \t %d \t %d\r\n", final_time, nav_data.val.t, t_delay);
}

```

mcprakash, 2018-01-10:
Attaching log output with your code
[putty.log](https://github.com/ARMmbed/sd-driver/files/1620544/putty.log)

deepikabhavnani, 2018-01-10:
t.start() should be done before reading the timer. See https://os.mbed.com/docs/v5.6/mbed-os-api-doxy/classmbed_1_1_timer.html

deepikabhavnani, 2018-01-10:
int final_time = 0;
while(1) {
t.start();
nav_data.val.t = t.read_ms();
sd_write_nav_data();
sd_write_imu_data();
final_time = t.read_ms();
t_delay = DT - (final_time - nav_data.val.t);
radio.printf("%d \t %d \t %d\r\n", final_time, nav_data.val.t, t_delay);
}

mcprakash, 2018-01-10:
Yes, it is declared outside of the while(1) loop

deepikabhavnani, 2018-01-10:
Can you reset the timer/start again inside the loop, to avoid overflow.

mcprakash, 2018-01-10:
Here you go
```
while(1) {
	t.reset();
	nav_data.val.t = t.read_ms();
	sd_write_nav_data();
	sd_write_imu_data();
	final_time = t.read_ms();
	t_delay = DT - (final_time - nav_data.val.t);
	radio.printf("%d \t %d \t %d\r\n", final_time, nav_data.val.t, t_delay);
	}
```
The log file is attached
[putty.log](https://github.com/ARMmbed/sd-driver/files/1620624/putty.log)

mcprakash, 2018-01-10:
Hi @deepikabhavnani 
 Please do let me know the next steps. 

Best
Chandra

deepikabhavnani, 2018-01-11:
Hello @mcprakash 
I created an example based on your test application, https://gist.github.com/deepikabhavnani/03af1eb0a34fadf4e245802b02b65fd8

In this if you see, 1024 bytes block of data takes (xxx)ms of time which varies based on device and is not less then 10 ms. 
In case of FAT filesystem, when you do writes smaller then a block size, if caches the data (which takes < 10 ms) and when one full block is available it performs the write (which is the case of > 10 ms) and you see actual write on SD.
If you have strict 10 ms window, I would recommend to have multi threaded application in which write is done by other thread else use event loop (10ms) and do the writing on SD card when you have event.
https://os.mbed.com/docs/v5.7/reference/event.html

mcprakash, 2018-01-13:
Hi @deepikabhavnani 
 Thank you so much for your assistance. You are awesome. 

 I am using buffers to store sensor data. Then when 1024 bytes are filled, I write to SD card using an Signal Event to a thread. Everything works well. 

 I have one final question. I am attaching the log file showing the timing delays. 
[putty.log](https://github.com/ARMmbed/sd-driver/files/1627879/putty.log)

 As seen, when the write thread is active, the cycle time reduces to 2 ms while it is 6ms (if write is not active). My question is how can I distribute this 4ms delay to 6 cycles rather than have 1 cycle among 6 cycles run slower than the rest.

 Best
 Chandra

deepikabhavnani, 2018-01-13:
I am not sure how to achieve that, even if I think of some logic to achieve even distribution it wont be same for all type of SD cards. To understand this try your timing test with Class4/Class 6/ and Class 10 cards.

mcprakash, 2018-01-13:
Hi @deepikabhavnani 
I had an idea, if a particular thread can be made to run only for a specific time (t_run = say 500 us), after which the priority is shifted to the main thread. I am not sure what functions/methods would enable such a feature, but it would be great if such a methodology exists. As for different cards, I can always measure the delay and distribute the timing accordingly.

closed 2018-01-16 by mcprakash