PelionIoT/sd-driver #81: Hardfaults occuring when using 5.6.1 sd-driver with IAR

edablan, 2018-01-12:
Calling ftell, and sometimes fopen causes the processor to hardfault. I have also noticed that when stepping through the disassembly and stepping into the mentioned file system functions, it is using the IAR compiled version of ftell and fopen. Is it possible that in my export of the project i missed a macro that prevents IAR from using its version of file system functions instead of using the mbed file system functions.

Also, the mbed OS version is 5.6.5 and the sd driver is 5.6.1 not sure if the pairing of these versions would cause any issues when I export. I was under the assumption that library versions are all backwards compatible. Any help would be much appreciated.

edablan, 2018-01-12:
I just tried this with a fresh new mbed project and fresh export all the while pairing mbed-os-5.6.1 and sd-driver-5.6.1 together and still have the same hard fault issues. Disassembly still shows the use of IAR specific file system functions.

edablan, 2018-01-15:
I just noticed that for remove the function being called is the defined mbed_retarget.cpp version, and not using an IAR specific version of remove. For some reason this still causes a hardfault for "An imprecise data access error"

deepikabhavnani, 2018-01-19:
> Also, the mbed OS version is 5.6.5 and the sd driver is 5.6.1 

SD driver version is still at 0.1.2. Latest tag created for SD driver is `sd-driver-0.1.2-mbed-os-5.6.1`.
Libraries are backward compatible, but some API change is implemented we either do a release or add version dependency in source code and tags.

As the tag name suggest sd-driver-0.1.2 version will be compatible to mbed-os-5.6.1 and higher.

deepikabhavnani, 2018-01-19:
> Calling ftell, remove, and sometimes fopen causes the processor to hardfault. I should also add, that this occurs with ftell not on the first call to it, but subsequent calls

Can you please let us know steps to reproduce or sample program to get this error.

edablan, 2018-01-24:
Ok so here is my write up on what the true issue is at least from what I have observed, and how things got fixed.

The ftell, fopen, and fclose issue were my fault and totally pilot error. I was inadvertently fclosing the file handle in other function and then trying to fclose it after the fact again. Totally dumb mistake.

However, the remove issue seems to be a bug with mbed-os-5.6.5. Here is my reasoning. I created a brand new project using mbed-os-5.6.5 that I cloned from git and added the lastest sd-driver. When I called remove and the file did not already exist on the storage medium,  remove would not gracefully error with -1 but instead hard fault. The key here to reproduce is the file cannot already exist in the storage medium when the call to remove is made. 

I then attempted this same test of creating new projects and adding sd-driver, but with newer versions of mbed os. I used mbed-os-5.7.0 and mbed-os.5.7.3, and had no hard faults when calling remove and the file not already existing on the storage medium. I did read that there were two known POSIX API bugs in mbed-os-5.6.5, just not specifically with remove. 

You might be wondering why not always build with the latest version, but we try to control our own source for the os we use to build with. That means subrepoing from what we deem as the blessed version and building from this version each time we create a new project. Obviously our new blessed version now will be the latest. Thanks for the response

deepikabhavnani, 2018-01-24:
@edablan - Glad issue was found and you got it working. I can understand not using the latest version always, but recently some good fixes were done with FAT along with ChanFS upgrade to resolve some more issues.

deepikabhavnani, 2018-01-24:
> I used mbed-os-5.7.0 and mbed-os.5.7.3, and had no hard faults when calling remove and the file not already existing on the storage medium.

Since issue is fixed already and I see your other issues are resolved as well, will close this. Thanks

closed 2018-01-24 by deepikabhavnani