PelionIoT/sd-driver #84: Seek test failing on K66F/GCC

marcuschangarm, 2018-02-23:
All other tests pass except for the seek test.

```
mbedgt: test suite report:
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
| target       | platform_name | test suite                      | result | elapsed_time (sec) | copy_method |
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | FAIL   | 65.13              | default     |
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+--------------+---------------+---------------------------------+----------------------------+--------+--------+---------+--------------------+
| target       | platform_name | test suite                      | test case                  | passed | failed | result  | elapsed_time (sec) |
+--------------+---------------+---------------------------------+----------------------------+--------+--------+---------+--------------------+
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Boundary seek and write    | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Large dir seek             | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Large file seek            | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Large file seek and write  | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Out-of-bounds seek         | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Seek tests                 | 1      | 0      | OK      | 49.48              |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Simple dir seek            | 0      | 1      | FAIL    | 0.06               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Simple file seek           | 0      | 0      | SKIPPED | 0.0                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek | Simple file seek and write | 0      | 0      | SKIPPED | 0.0                |
+--------------+---------------+---------------------------------+----------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 1 FAIL / 7 SKIPPED / 1 OK
mbedgt: completed in 65.32 sec
mbedgt: exited with code 1
```

deepikabhavnani, 2018-03-14:
+--------------+---------------+-------------------------------------+----------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                          | test case                                                                              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+-------------------------------------+----------------------------------------------------------------------------------------+--------+--------+--------+--------------------+
| K66F-GCC_ARM | K66F          | sd-driver-tests-block_device-basic  | Testing read write random blocks                                                       | 1      | 0      | OK     | 6.94               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_00: fopen()/fgetc()/fprintf()/fclose() test.                          | 1      | 0      | OK     | 0.17               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_01: fopen()/fseek()/fclose() test.                                    | 1      | 0      | OK     | 0.25               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_03: tmpnam() test.                                                    | 1      | 0      | OK     | 0.07               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_04: fileno() test.                                                    | 1      | 0      | OK     | 0.09               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_05: opendir() basic test.                                             | 1      | 0      | OK     | 0.65               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_06: fread()/fwrite() file to sdcard.                                  | 1      | 0      | OK     | 0.14               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_07: sdcard fwrite() file test.                                        | 1      | 0      | OK     | 0.17               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_08: FATFileSystem::read()/write() test.                               | 1      | 0      | OK     | 4.3                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_09: POSIX FILE API fread()/fwrite() test.                             | 1      | 0      | OK     | 4.86               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-basic    | FSFAT_BASIC_TEST_10: ChanFS read()/write()) test.                                      | 1      | 0      | OK     | 4.15               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory creation                                                                     | 1      | 0      | OK     | 0.41               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory failures                                                                     | 1      | 0      | OK     | 0.08               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory iteration                                                                    | 1      | 0      | OK     | 0.07               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory remove                                                                       | 1      | 0      | OK     | 0.33               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory rename                                                                       | 1      | 0      | OK     | 3.02               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Directory tests                                                                        | 1      | 0      | OK     | 9.16               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | File creation                                                                          | 1      | 0      | OK     | 0.08               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Multi-block directory                                                                  | 1      | 0      | OK     | 50.3               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Nested directories                                                                     | 1      | 0      | OK     | 1.13               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-dirs     | Root directory                                                                         | 1      | 0      | OK     | 0.07               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-files    | Block Size file test                                                                   | 1      | 0      | OK     | 0.53               |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-files    | Dir check                                                                              | 1      | 0      | OK     | 0.06               |
[... cut ...]
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek     | Simple file seek                                                                       | 1      | 0      | OK     | 0.1                |
| K66F-GCC_ARM | K66F          | sd-driver-tests-filesystem-seek     | Simple file seek and write                                                             | 1      | 0      | OK     | 0.12               |
+--------------+---------------+-------------------------------------+----------------------------------------------------------------------------------------+--------+--------+--------+--------------------+

deepikabhavnani, 2018-03-14:
@marcuschangarm  - All the test on K66F were successful with sd-driver master and  mbed-os-5.7. 
Can you help in reproducing the failure (mbed-os version / sd-driver version)

marcuschangarm, 2018-03-14:
I just re-ran the test using:

sd-driver: 10aac9f16207e81d0f231dfc9eef7ac88ad32c27
mbed-os: 91e6db1ea251ffcc973001ed90477f42fdca5751

same result.

marcuschangarm, 2018-03-14:
Fix incoming: https://github.com/ARMmbed/mbed-os/pull/6363

closed 2018-03-14 by marcuschangarm