PelionIoT/sd-driver #87: How do you handle card removal and reinsertion?

GrahamSayers, 2018-03-17:
Hi Simon,

Thanks for your work on this library.  I'm trying to use this in a project.  It's mostly working, but I'm struggling with card removal and reinsertion.  I'm using STM32L476VG and mbed 5.4.2 and sd-driver-0.0.3-mbed-os-5.4.1 and gcc 4.9.  I have been running example1.cpp and variations.  I want to use the posix file api and write text files to the card.  My test is to call the body of example1.cpp in a loop, once every 5 seconds or so.

As long as the card is inserted it works but when I remove the card it often hangs.  The rtos thread it's in is actually blocked sometimes for 60 seconds or more - sometimes it resumes and sometimes it doesn't.

What is the right way to test for or deal with card removal?  Looking through various examples, the approach seems to be to try to open a file and check if the return handle is NULL.

FILE* fp = fopen(filename, mode);

Checking for NULL sometimes works.  But sometimes even if the card is removed fp is not returned NULL.  Also, if the card is removed, sometimes the test seems to hang inside the call to fopen().  I built the test in the online compiler using same mbed and sd-driver version and the same thing happens.  Also tried 5.4.1 with the same result.

So my question is, what is the right way to detect and handle card removal and reinsertion?  Is there something I should be doing when I encounter the first error to reset the state?  

Thanks!

geky, 2018-03-17:
Hi Graham,

Thanks for raising the issue.

Does your hardware have a switch that connects when the SD card is inserted? Often times a lot of them do, and in that case the easiest thing to do is connect an interrupt to the pin or read it's value when you need to.

Right now, I don't think the driver is written to handle a removed card. It probably should return an error on any block device operation if the card isn't present, and this should be propagated up through the filesystem as EIO.

After reinsertion you would need to unmount and remount the filesystem, there's just too much state involved. Additionally the user may have modified the filesystem in the meantime.

Unfortunately I don't know when we'll be able to look into this, but would love to take in a patch if you get something working.

cc @deepikabhavnani thoughts?

GrahamSayers, 2018-03-17:
Thanks for fast response.  The card holder does have a card detect pin.  Unfortunately we were short on micro pins and I told the PCB guy not to connect it - I hadn't seen any examples using that pin.  I'll put a bodge wire on and try it.  If it works we can fix it on the next rev.  I'll poke at the driver some more but it's large stack and I am not familiar with it so I don't have a much hope.

Is there a right way to mount and unmount the filesystem?  Mount obviously is called in FATFileSystem ctor and dtor.  Can you just do  fs.mount();   wherever you want?

Can I just wrap my card access in mount() and unmount() calls to ensure it is reset every time?

Thanks.

closed 2018-03-17 by GrahamSayers