PelionIoT/sd-driver #93: SD card init fails randomly

JaniSuonpera, 2018-04-19:
I have detected that SD card init fail randomly with K64F and https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/source/platform/mbed-os/common_setup.cpp#L296-L303. In this case I get  SD_BLOCK_DEVICE_ERROR_NO_DEVICE https://github.com/ARMmbed/sd-driver/blob/f9c33caa3252792fdd4b8a421367027be3cf6b0e/SDBlockDevice.cpp#L176 even SD card has been connected. 

```
...
08:29:13.488 | D1 <<< DutThread: Application ready. Build at: Apr 19 2018 08:07:11
08:29:13.488 | D1 <<< DutThread: Mbed OS version 5.8.2
08:29:13.488 | D1 <<< DutThread: mcc_platform_storage_init() - bd->init() failed with -5005
08:29:13.488 | D1 <<< DutThread: Failed to initialize storage
08:29:13.489 | D1 <<< DutThread: Initialization failed, exiting application!
...
```
What is doable in this case?

deepikabhavnani, 2018-04-19:
@JaniSuonpera I will need some more information related to this issue:
1. Is SD card init successful on subsequent tries or once fail it always fails? 
2. Which version of SD driver is used (SHA)?
3. Card details (Vendor / ID/ Size)

JaniSuonpera, 2018-04-20:
@deepikabhavnani 

1. It is tried init once during startup. Should there be retry if init fails?
2. ae7e7440054c9447f8255bdccbcc523b3f6dffe4 , we are using sd-driver though https://github.com/ARMmbed/storage-selector
3. scandisk-ultra-16G

deepikabhavnani, 2018-04-20:
No, retry wont help much as init fails once might need reset for the card.
I wanted to know if the card is in usable state? If we plug it out and put it back in does that work?
Do we have any particular scenario in which the card is failing? As I guess from random is that it works for few runs and then fails initialization.

deepikabhavnani, 2018-04-20:
Also wanted to confirm init issue is only with K64F?

petrai01, 2018-04-23:
I have also seen this relatively often with random K64Fs in our RAAS setup. Memory card type doesn't seem to be a major factor in this, I've seen this happen with scandisk-ultra-16G equiped devices, and those that have smaller random cards:
https://jenkins-internal.mbed.com/job/sda_regression_integration/437/artifact/log/2018-04-20_035458.917287/create_and_test_new_access_policy_0/Dut.D1.log
https://jenkins-internal.mbed.com/job/sda_regression_os2/303/artifact/log/2018-04-19_125242.245596/try_to_set_value_with_expired_token_0/Dut.D1.log

JaniSuonpera, 2018-04-23:
We also using RAAS setup and we are mostly using K64F.  I cannot say that is only with K64F. There are some logs.
```
...
06:19:15.411 | D1 <<< DutThread: Application ready. Build at: Apr 23 2018 06:06:46
06:19:15.411 | D1 <<< DutThread: Mbed OS version 5.8.2
06:19:15.411 | D1 <<< DutThread: No response CMD:0
06:19:15.411 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0 response: 0xff
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0 response: 0xff
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.412 | D1 <<< DutThread: No response CMD:0
06:19:15.413 | D1 <<< DutThread: No response CMD:0 response: 0xff
06:19:15.413 | D1 <<< DutThread: No response CMD:0
06:19:15.511 | D1 <<< DutThread: No response CMD:0
06:19:15.511 | D1 <<< DutThread: No response CMD:0
06:19:15.511 | D1 <<< DutThread: No response CMD:0 response: 0xff
06:19:15.511 | D1 <<< DutThread: No response CMD:0
06:19:15.511 | D1 <<< DutThread: No response CMD:0
06:19:15.511 | D1 <<< DutThread: No response CMD:0
06:19:15.512 | D1 <<< DutThread: No response CMD:0 response: 0xff
06:19:15.512 | D1 <<< DutThread: No disk, or could not put SD card in to SPI idle state
06:19:15.512 | D1 <<< DutThread: Fail to initialize card
06:19:15.512 | D1 <<< DutThread: mcc_platform_storage_init() - bd->init() failed with -5005
...
```

deepikabhavnani, 2018-04-23:
This behaviour should be linked to any previous failures. Do you see any failure (read/write/erase) in filesystem?

JaniSuonpera, 2018-04-24:
@geky Do you know some filesystem failure, which might cause this?

JaniSuonpera, 2018-04-24:
This seems to be only K64F issue.  Failure count has been increased with mbed-os 5.8.2.

deepikabhavnani, 2018-04-24:
Please check Mbed OS 5.8.2 release notes for any change in SPI driver or SPI of target (k64f)

geky, 2018-04-24:
I was looking through this earlier, but I have no idea why the failure is happening. It's erroring before the filesystem so it's not the result of a filesystem issue.

teetak01, 2018-04-24:
This is really rare event and very difficult reproduce. I observed it to occur with four different boards in CI. I then took control of all four, but no matter what I tried, it always worked fine.

petrai01, 2018-04-24:
Maybe the way RAAS works makes this happen more often in CI. Here you can see every other test run fail due to this issue:
https://jenkins-internal.mbed.com/job/sda_regression_os2/

deepikabhavnani, 2018-04-24:
@petrai01 - please explain the failure scenario. Is it
1. Test 1 success - test 2 init fail - test 3 success (without any human interference) 
2. Test 1 success - test 2 fail (not init any failure) - test 3 init fail - test 4 init fail and all subsequent test init fails
3. Test 1 success - test 2 init fail - all subsequent test after this fail in init and need to remove card and reset device to make it work.   

Want to know the failure pattern is it from any of the above cases?

deepikabhavnani, 2018-04-24:
If it is option , it means the card is fine and we see init timing issue in SD after reset. We will need SPI bus trace of failing case, else I guess failure can be reproduced by running read/write test with resets. 

I will be good if some one can capture trace for failing case, else I will do it after 2 days (I am out of office)

JaniSuonpera, 2018-04-24:
It is also option 1. I have tried to reproduce issue but I didn't succeed. I am seeing this with RaaS devices. I didn't succeed to rerpoduce this locally with my own K64F.

deepikabhavnani, 2018-04-24:
It looks like timing issue, can we try init 2/3 times to see if that works. ( not proposing that as solution but to understand issue)

petrai01, 2018-04-25:
I'm not sure if this is anyway helpful, but this particular K64F seems to be prone quite to this error:
http://ropi.mbedcloudtesting.com:8000/resource/0240000032044e45001d7009997b001c6781000097969900

https://jenkins-internal.mbed.com/job/sda_regression_prod2/290/artifact/log/2018-04-23_201615.653862/result.html

petrai01, 2018-04-26:
@deepikabhavnani - the DUT is reset between retries

deepikabhavnani, 2018-07-05:
Issues was resolved as per internal jira ticket IOTMORF-2327.

closed 2018-07-05 by deepikabhavnani