PelionIoT/sd-driver #99: How to truncate

OscarGarciaF, 2018-05-24:
How can i truncate a file to a certain size with this library?

geky, 2018-05-24:
Hi @OscarGarciaF, unfortunately we currently don't have support for the `truncate` and `ftruncate` functions. Your best option is to create a new file and copy the contents until it reaches the new size you want.

This is a missing part of the Mbed APIs, but most of the file system implementations do have an implementation of truncate. Poking the API through has been low priority since no one has asked for it.

I copied this issue over to the mbed-os repo where it will have more visibility. We'll triage it and see when we can add the API:
https://github.com/ARMmbed/mbed-os/issues/7010

If it's a high priority for you, you're welcome to add the API yourself to the mbed-os repo. Out of curiousity which filesystem are you using?

OscarGarciaF, 2018-05-24:
Im using the FATFileSystem, I want to implement a datalogger, but with files in the order of GBs i dont want to be copying new files everytime I process a line of the .csv file.
I have tried to use the LittleFileSystem that has a truncate function but it does not seems to work (fopen and stat fail where FATFileSystem does not)

closed 2018-11-06 by deepikabhavnani