PelionIoT/sd-driver #100: Multiple SPIs in the same channel

OscarGarciaF, 2018-05-31:
Im trying to use this library while simultaneously using an ethernet shiled connected to the same SPI channel with their respectives CS. While they work separately, neither work while the other is present.

Is there a special way this should be done?

More info: ARMmbed/#7064

deepikabhavnani, 2018-05-31:
Please share details of target device and Ethernet shield.

OscarGarciaF, 2018-05-31:
NUCLEO-F103RB board
enc28j60 module

I don't think it's a hardware problem, i just wanted to know what's the proper way to handle two SPIs on the same channel. 
Is it ok to have two SPIs instances or should it be just one?
Can they work at different frecuencies?

deepikabhavnani, 2018-05-31:
Separate instances of SPI will be required for each interface. As part of write, we do update the frequency and format before each write for that specific instance hence it should work with different frequencies.

https://github.com/ARMmbed/mbed-os/blob/master/drivers/SPI.cpp#L102

Please note in `_acquire()` https://github.com/ARMmbed/mbed-os/blob/master/drivers/SPI.cpp#L94, private class members are used to set width, frequency  (set by constructor or frequency()/format() call)
before access to SPI bus, hence separate driver instance will be needed.

OscarGarciaF, 2018-05-31:
@deepikabhavnani then i have no idea why it's not working.
If both spi instances have their respective CS pin delcared, do i need to select and deselect them manually or just the "other" CS?
For example giveng two spi instances: A and B
when using A, do i only need to manually deselect B or does the library for B deselect itself when it's done using the channel?
(take in mind i'm only talking about the sd-driver class)

deepikabhavnani, 2018-05-31:
@OscarGarciaF - I don't think CS switching should be needed by application it should be handled by SPI driver. 
Can you confirm CS is not done by tracing SPI bus signal ?

deepikabhavnani, 2018-05-31:
As part of sd-driver CS select/deselect is done as part `SDBlockDevice::_select()` and `SDBlockDevice::_deselect()` calls.

OscarGarciaF, 2018-05-31:
Checking the other library i can confirm it also selects and deselects its CS, it's hard to verify the bus since i'm not using a breadboard.
But if both of them work on their own i guess they are selecting and deselecting appropriately.

It's a weird case since the ethernet fails before even calling SDBlockDevice.init(), the mere prescence of the instance makes it fail

deepikabhavnani, 2018-05-31:
In SPI driver, `__HAL_SPI_DISABLE` and `__HAL_SPI_ENABLE` is done as part of init. Could that be part of problem? Some investigation is needed here I guess.

Also, as per the comments "https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/stm_spi_api.c#L90" we can have SPI enabled all time for standard 4 wires SPI, but not sure how that is achieved.

OscarGarciaF, 2018-05-31:
I'm not sure what the problem is since i don't know how the SPI API works, in any case even after both of my SPIs call their init function they still don't work.
I already posted the issues on the mbed-os repo:
https://github.com/ARMmbed/mbed-os/issues/7064

OscarGarciaF, 2018-05-31:
@deepikabhavnani would it matter if the ethernet SPI is being used on a Thread by the RTOS?

OscarGarciaF, 2018-05-31:
Insufficent stack size of the thread made it hang. I guess since one hanged it made the SD spi fail too.
I allocated more memory and it works now, thanks for the help

closed 2018-05-31 by OscarGarciaF