PelionIoT/sd-driver #101: Llittlefs on sd-driver

OscarGarciaF, 2018-06-01:
does sd-driver  support littlefs?
I have tried using it with File and FILE and it does no seems to work.
Also stat() seems to always fail.

Is there some special way to accomplish this?

geky, 2018-06-04:
Yes is should work!

Have you tried starting with our filesystem example as a baseline?
https://github.com/armmbed/mbed-os-example-filesystem

Were you able to use the FAT filesystem on the SD card? Do you know if there might be hardware problems?

One thing to note, FAT and littlefs are inherently different filesystems on the SD card, so you do need to reformat the SD card in order to switch between the two.

If you're using logic that trys to mount, and falls back to format, it's possible formatting one filesystem corrupted the other in a way that mount can't detect. You can try calling littlefs's format directly before using the SD card. (We have a fix for this in mbed-os-5.9, but it hasn't been released quite yet).

OscarGarciaF, 2018-06-04:
it worked on the FAT, i will try formatting to see if it works

OscarGarciaF, 2018-06-08:
format (i have tried calling it before mount) and .mount both hang and never return a value

geky, 2018-06-12:
That's unfortunate. Some questions:
- What board are you using?
- What mbed-os version? (`git log`)
- What sd-driver version? (`git log`)
- Are you able to share any code that shows the issue?

I'm not aware of any issues with SD and littlefs, I'm not sure what the issue could be.

JanneKiiskila, 2018-09-12:
It definitely should work, it works with the Mbed Cloud Client.

closed 2018-11-06 by deepikabhavnani